osrf/lrauv #92: YoYo expectations flaky.

arjo129, 2021-11-25:
Recent runs of the YoYo test seem to be flaky across all branches and PRs. The test occasionally fails but passes after a rerun.

chapulina, 2021-11-25:
Since the expectations were changed in #76, we've had only one failure on `main`:

```
2021-11-25T03:22:31.2937683Z 8: /home/developer/lrauv_ws/src/lrauv/lrauv_ignition_plugins/test/test_mission_yoyo_circle.cc:100: Failure
2021-11-25T03:22:31.2941970Z 8: Expected: (-22.4) < (pose.Pos().Z()), actual: -22.4 vs -22.4337
2021-11-25T03:22:31.2944353Z 8: 8200
2021-11-25T03:22:31.2948272Z 8: /home/developer/lrauv_ws/src/lrauv/lrauv_ignition_plugins/test/test_mission_yoyo_circle.cc:100: Failure
2021-11-25T03:22:31.2953717Z 8: Expected: (-22.4) < (pose.Pos().Z()), actual: -22.4 vs -22.4432
2021-11-25T03:22:31.2958430Z 8: 8300
2021-11-25T03:22:31.2962710Z 8: /home/developer/lrauv_ws/src/lrauv/lrauv_ignition_plugins/test/test_mission_yoyo_circle.cc:100: Failure
2021-11-25T03:22:31.2972449Z 8: Expected: (-22.4) < (pose.Pos().Z()), actual: -22.4 vs -22.445
2021-11-25T03:22:31.2976574Z 8: 8400
2021-11-25T03:22:31.2981772Z 8: /home/developer/lrauv_ws/src/lrauv/lrauv_ignition_plugins/test/test_mission_yoyo_circle.cc:100: Failure
2021-11-25T03:22:31.2988120Z 8: Expected: (-22.4) < (pose.Pos().Z()), actual: -22.4 vs -22.4282
2021-11-25T03:22:31.2988925Z 8: 8500
```

That means the vehicle is going deeper than expected, in theory it shouldn't go deeper than `-20`. We can increase the tolerance by 0.1m...

arjo129, 2022-01-08:
Fixed by #133

closed 2022-01-08 by arjo129