osrf/lrauv #144: Add AHRS sensor

chapulina, 2022-01-12:
Add an "Attitude and Heading Reference System", which we may be able to implement as a combination of an IMU and a magnetometer.

Here's the data it needs to output:

```
float roll: rad in NED world frame
float pitch: rad in NED world frame
float yaw: rad, vehicle heading relative to true north (in NED)

float accelX, accelY, accelZ: m/s2, vehicle-frame linear accelerations in XYZ (XYZ=FSK)
float gyroX, gyroY, gyroZ: rad/s, vehicle-frame angular velocities in XYZ (XYZ=FSK)

// Optional (not really required)
float magX, magY, magZ: microtesla (can also be in milligauss), vehicle frame magnetometer reading in XYZ (XYZ=FSK)
```

hidmic, 2022-02-03:
@caguero @braanan I've been digging through LRAUV and Ignition code. Also played a bit with the latest MBARI Docker image. I think we have two options here:

  1. We arrange an AHRS by adding stock [IMU](https://github.com/ignitionrobotics/ign-sensors/blob/main/src/ImuSensor.cc) and [magnetometer](https://github.com/ignitionrobotics/ign-sensors/blob/main/src/MagnetometerSensor.cc) sensors to the `tethys_equipped` SDF model. They seem feature complete, offer a typical `ignition-transport` interface using stock [`imu.proto`](https://github.com/ignitionrobotics/ign-msgs/blob/main/proto/ignition/msgs/imu.proto) and [`magnetometer.proto`](https://github.com/ignitionrobotics/ign-msgs/blob/main/proto/ignition/msgs/magnetometer.proto) messages, and IIUC we can arrange them so that their local frame matches FSK. If we don't need data synchronization between both sensors and nobody has strong opinions against stock interfaces, this is perhaps our best option.
2. We merge (and perhaps simplify?) stock IMU and magnetometer sensors' implementation into a single AHRS plugin. We'll need an `ahrs.proto` message that combines both `imu.proto` and `magnetometer.proto` messages. This one's perhaps the most flexible if we have some funky needs.

In both cases, we'll need to update the `TethysComms` plugin to populate an extended `lrauv_state.proto` message if we want to maintain compatibility. Thoughts?

caguero, 2022-02-07:
We could start with (1) that requires less custom code. We could potentially play with the topic names to give the impression that data is coming from the same sensor. 

`/model/tethys/ahrs/imu`
`/model/thetys/ahrs/magnetometer`

braanan, 2022-02-07:
Thanks, Mich. I agree with @caguero, option 1 sounds like the path of least resistance. I think we should consider integrating the IMU/Mag receive interface in the lrauv-app instead of adding more fields to TethysComms' state msg. This should be straightforward as we already have multiple Ign transport nodes in the lrauv-app.

hidmic, 2022-03-23:
Alright, considering all PRs have been merged and Docker images have been updated, I think we call this done. Closing.

closed 2022-03-23 by hidmic