osrf/lrauv #149: Surface winds

caguero, 2022-01-19:
Add Ignition world plugin for simulation of surface winds (collaboration with Hamilton’s DOE).

chapulina, 2022-01-19:
One option is to use the existing [WindEffects](https://ignitionrobotics.org/api/gazebo/6.4/classignition_1_1gazebo_1_1systems_1_1WindEffects.html) plugin. It applies a sinusoidal force that is configurable at runtime through a topic. It's a world plugin, so it applies to all models in the world equally.

hidmic, 2022-02-17:
BTW `ignition::math` already implements a Catmull-Rom [spline](https://github.com/ignitionrobotics/ign-math/blob/ign-math6/include/ignition/math/Spline.hh).

caguero, 2022-02-21:
It sounds like a good plan to me.

caguero, 2022-02-21:
My only thought is to think about how complex is to make the function depend on `x`, `y` and `z`. That way it'll be a bit more generic for the future.

hidmic, 2022-02-21:
@caguero as we talked earlier, I think a simple yet generic way to make  `forceApproximatingScalingFactor` a function of `x`, `y`, and `z` would be to define it as `f(x, y, z) = p(x).q(y).r(z)` where `p`, `q`, and `r` are splines (e.g. piecewise cubic polynomials). Voxel grids and simple surfaces would be easy to define. Describing more general surfaces in those terms may be challenging though.

hidmic, 2022-03-09:
@caguero @arjo129 @braanan see quick demo for surface winds effects below:

https://user-images.githubusercontent.com/13500507/157509218-a17b622a-8b29-4ea8-9ae9-050f1804beec.mp4

I used the `buoyant_tethys.sdf` world as a basis. At first, you can see the vehicle getting pushed East by the wind. The vehicle then submerges and cruises downwards, then West, and then upwards. When it emerges, it starts getting pushed East by the wind again.

braanan, 2022-03-10:
Thanks, @hidmic. Good progress. Do you happen to know if the vehicle was set to be positively buoyant in that run? Also, what wast the wind magnitude in that run? 

Thanks

hidmic, 2022-03-15:
Argh, I thought I had replied. 

> Do you happen to know if the vehicle was set to be positively buoyant in that run?

I used `buoyant_tethys.sdf` defaults, and I did not deflate nor inflate the buoyancy engine from its neutral volume so perhaps it was positively buoyant (a sensible default?).

> Also, what wast the wind magnitude in that run?

1 m/s, East.

hidmic, 2022-03-16:
@braanan as per f2f chat, I'll check what's going on with hydrodynamic drag here. Vehicle is moving too fast for such calm wind conditions.

hidmic, 2022-03-23:
@braanan there's a non-zero wrench due to hydrodynamic drag when winds are pushing the vehicle, so we're good on that front. I think it comes down to force approximation:

https://latex.codecogs.com/png.image?\dpi{110}F_w&space;=&space;s_f&space;m_b&space;(v_w&space;-&space;v_b)

In the short video above, I used a unit scaling factor above sea water level. Hydrodynamic drag forces are high as a result, but still, the vehicle moves fast. Reducing the scaling factor, e.g. to 0.1, gets us closer to the slow, expected motion. With such a simple approximation, we'll have to tweak it quite a bit.

PR #179 (2022-05-04): Add above surface winds
Closes #149. This patch requires https://github.com/ignitionrobotics/ign-gazebo/pull/1357.
closed 2022-05-04 by arjo129