osrf/lrauv #159: Investigate why test fixtures go :boom: if we have more than one TEST in a single process.

arjo129, 2022-02-16:
Google protobuf doesn't like the way we are linking the messages in our repository. Whenever we try to run two TestFixtures in the same process the tests will segfault.

hidmic, 2022-05-19:
https://github.com/protocolbuffers/protobuf/issues/4126 suggests loading/unloading protobuf message shared libraries leaves bad global state behind. Perhaps we can preload them?

arjo129, 2022-09-08:
Hopefully eventually we can move away from custom messages then?

marcoag, 2022-09-08:
Dunno if related but I also had troubles trying to run two servers in the same test in `gz-sim` so I have to split them in two files ([here](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/test/integration/model_photo_shoot_default_joints.cc) and [here](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/test/integration/model_photo_shoot_random_joints.cc)).

hidmic, 2022-11-10:
We know why this happens and it's very much core to how `protobuf` works. I'm inclined to close this as `wont fix`.