osrf/lrauv #164: Unable to find image 'mbari_lrauv:latest' locally

romulogcerqueira, 2022-03-04:
```pre
docker: Error response from daemon: pull access denied for mbari_lrauv, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
```

arjo129, 2022-03-07:
Hey @romulogcerqueira ,
Which command did you use to start the simulator? This seems like a docker issue more than anything with the sim itself.

romulogcerqueira, 2022-03-07:
Hi @arjo129 ,

the error happens when I run `./build_and_run_docker.sh`

```pre
romulo@N8xxEP6:docker (main)$ ./build_and_run_docker.sh 
xauth:  file /tmp/.docker.xauth does not exist
Sending build context to Docker daemon  71.94MB
Error response from daemon: dockerfile parse error line 52: unknown instruction: &&
Unable to find image 'mbari_lrauv:latest' locally
docker: Error response from daemon: pull access denied for mbari_lrauv, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.

```

arjo129, 2022-03-07:
Can you pull the latest version again it would be fixed by #84

romulogcerqueira, 2022-03-07:
Hi @arjo129, thanks for your quick reply and congratulations for this project.

I was able to run the simulation by running `./docker/debug_integration.sh`. The file `./docker/debug_and_run.sh` does not exist.

closed 2022-03-07 by romulogcerqueira