osrf/lrauv #184: Plot 3D path in empty_environment for spawned vehicle

braanan, 2022-03-29:
Hi @chapulina, 

Is there a way to add Plot3D for spawned vehicles in empty_environment.sdf? Guess one way to do this would be to hardcode all LRAUV names into the sdf... Any thoughts?

Thanks!

chapulina, 2022-03-29:
Humm interesting use case. We currently have `tethys` hardcoded there:

https://github.com/osrf/lrauv/blob/861e7f2fc7fccf8ea5c1bd0c573227b355648e57/lrauv_ignition_plugins/worlds/empty_environment.sdf#L246

`Plot3D` can be instantiated multiple times, each attached to a different entity and with different configurations. So right now it's possible to manually insert it:

1. Open the empty world
2. Spawn the vehicles
3. From the top-right menu, insert a `Plot 3D` plugin
4. Also insert an `Entity Tree` (we may want to hardcode this one on every world, it's very handy)
5. On the entity tree, select the vehicle you want to plot. Its name should show up on the Plot 3D and it should start plotting
6. You can click the lock icon on Plot 3D so the entity doesn't change when you click away from it

To do it more automatically, I think we'd need:

1. The ability of loading GUI plugins using transport: https://github.com/ignitionrobotics/ign-gui/issues/379
2. Then we use that on `WorldCommPlugin` after we spawn each vehicle.

braanan, 2022-03-30:
Ok, thanks for looking into it — I'll give the manual approach a try.

closed 2022-12-26 by caguero