osrf/lrauv #201: Model battery discharge dynamics

hidmic, 2022-04-26:
## Preliminaries

As first step, use upstream [`LinearBatteryPlugin`](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/systems/battery_plugin/LinearBatteryPlugin.cc) plugin to model battery discharge at a constant power load.

caguero, 2022-04-27:
> ## Preliminaries
> As first step, use upstream [`LinearBatteryPlugin`](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/systems/battery_plugin/LinearBatteryPlugin.cc) plugin to model battery discharge at a constant power load.

braanan, 2022-05-03:
LRAUV battery pack is made of 62 Inspired Energy rechargeable smart Lithium-Ion battery sticks (model#: [NH2034HD34](https://github.com/osrf/lrauv/files/8614587/NH2034HD34_spec_v1.5.pdf)). Each stick packs 6.27 A-hrs at 14.4 volts, with a nominal voltage range of 16.8V to 11V. At full charge, the LRAUV battery capacity is 400 A-hr.

LRAUV nominal discharge is 2 Amps while underway at 1 m/s with the science payload activated. We can break up power usage more granularly per sub-system as needed. 

[NH2034HD34_spec_v1.5.pdf](https://github.com/osrf/lrauv/files/8614587/NH2034HD34_spec_v1.5.pdf)

closed 2022-07-07 by caguero