osrf/lrauv #224: Documentation landing page

caguero, 2022-07-07:
Design a landing page in GazeboSim to be used as a generic landing page for anyone interested in maritime simulation. This landing page should capture what we can do, limitations, worlds, models, plugins, etc. It should contain links to relevant tutorials.

We should also design a set of tutorial that could live in:
 * GazeboSim: General-purpose tutorials.
 * LRAUV project: Specific tutorials to this project.

In the LRAUV wiki we should create a complete story for anyone interested in this project (specially from MBARI). It's OK to point to a tutorial in GazeboSim if needed.

Some ideas to cover:
  * Installation
  * API
  * Examples
  * Architectural overview
  * Message definitions
  * Configuration
  * How to generate new science data?
  * How to generate new DEMs?

caguero, 2022-08-03:
Plan for testing

Getting started: Mich
Simulation intrinsics: Arjo
Data generation: Aditya

adityapande-1995, 2022-08-08:
For data generation, it would be nice to have a short set of commands (more like a tutorial) that one can run to actually see the data in action. For eg, for someone new, this might be a helpful way:

## Usage

### The world file
(We can explain how the world file should look like here, maybe the plugins and models it must contain, to have a bare minimum example working)
Example science data csv files can be found here ``<link>``. We can use ``<this>`` world file to use that science data, which uses ``<this>`` plugin to to load and publish the data. The path for the science data should be specified in the world file as follows (path should be relative to what ?):

```
<!-- Example world file --->
.
.
<world>
  <plugin name="SomeName" 
   csv_path="/path/to/csv">
   .
   .
 <include some_model>
  .
</world>
```

### Read the published data

The world file described above is just a template. We'll use ``this`` file to run an example, which contains data from ``this`` csv file.

```
gz sim -v 4 -s some_science_data_demo_world.sdf``
```

(I'm not sure how the data can be echoed or introspected, but if its a topic, that would be nice.)
We can check the generated data using : 
``` 
gz topic -e -t /science/data/topic
```

### Another way to load csv files

We can also load the csv file using the topic [/world/science_sensor/environment_data_path topic](https://github.com/osrf/lrauv/wiki/API#science-sensors-system). For e.g, we can run a black world that does not contain the data path in the plugin: 

```
 gz sim world_with _no_csv_path.sdf
```
We can set the path using : 
```
gz topic -p -t /world/science_sensor/enviroment_data_path_topic ...
```

We can see the changes are reflected using : 
``` 
gz topic -e -t /science/data/topic
```

marcoag, 2022-08-11:
Thanks for the feedback @adityapande-1995, just added the tutorial [here](https://github.com/osrf/lrauv/wiki/How-to-generate-new-science-data%3F#usage).

hidmic, 2022-08-13:
@marcoag great writing! A few observations:

- There's quite a few places that still refer to `Ignition`. As we sort out #234, maybe we should revisit these pages.
- On [`Test integration with MBARI LRAUV code base`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#run-the-mbari-lrauv-application), consider:

```diff
-Time synchronization has been done between the MBARI application and the Ignition simulation, such that in most cases, the controller and the simulation should be running in synchronization. There may be corner cases that still need to be resolved.
+The MBARI application relies on the Gazebo simulated clock for timekeeping, such that in most cases, both controller and simulation should be synchronized. There may be corner cases that still need to be resolved.
```

Also, what you mean by corner cases?

- Also on  [`Test integration with MBARI LRAUV code base`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#run-the-mbari-lrauv-application), it reads:

> On the vehicle, an app is always being run. If no missions are specified, then it is running the default. On the real vehicle, the default mission is GoToSurface. (NOTE: we have removed the default app in the MBARI public image until [this issue](https://github.com/osrf/lrauv/issues/38) is resolved. Currently, nothing is being run by default. Skip this check.) Verify that it is running the default GoToSurface app:

Which default app?

- On [`How to Send and Receive Data Packets on the Acoustic Subsystem.`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#direct-method), where's the `acoustic_comms_test.sdf` world?
- Also on [`How to Send and Receive Data Packets on the Acoustic Subsystem.`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#direct-method), I think we could just include the example and compile it with the rest.

marcoag, 2022-08-15:
@hidmic thanks for the feedback!

>     * There's quite a few places that still refer to `Ignition`. As we sort out [Ign to Gz migration #234](https://github.com/osrf/lrauv/issues/234), maybe we should revisit these pages.

Good point, I will add a list of places to review to [Ign to Gz migration #234](https://github.com/osrf/lrauv/issues/234) so we keep it in mind.

> * On [`Test integration with MBARI LRAUV code base`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#run-the-mbari-lrauv-application), consider:
>
> ```diff
> -Time synchronization has been [...]

Added the change, thanks!

> Also, what you mean by corner cases?

No idea tbh, this is a direct copy of the current [README.md](https://github.com/osrf/lrauv/blob/main/README.md#run-the-mbari-lrauv-application), according to [this](https://github.com/osrf/lrauv/blame/main/README.md) maybe @mabelzhang knows something? @arjo129 maybe?

> * Also on  [`Test integration with MBARI LRAUV code base`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#run-the-mbari-lrauv-application), it reads:
>
>
>
> Which default app?

Also a direct copy from the `README.md`. I believe this is the default mission (shouldn't be called app?) on the MBARI code, on top of that that [issue](https://github.com/osrf/lrauv/issues/38) seems solved already, @arjo129 do we know the status of this?

>     * On [`How to Send and Receive Data Packets on the Acoustic Subsystem.`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#direct-method), where's the `acoustic_comms_test.sdf` world?

Also remainder from `README.md` copy/paste, it seems removed long time ago and that the replacement is ` acoustic_comms_fixture.sdf`, will update the docs accordingly.

>     * Also on [`How to Send and Receive Data Packets on the Acoustic Subsystem.`](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials#direct-method), I think we could just include the example and compile it with the rest.

Good point, will follow up with a PR that adds this to the examples.

mabelzhang, 2022-08-15:
Arjo probably has the latest. But I will respond what I meant when I wrote that.

> Which default app?

It's referring to GoToSurface. The MBARI folks called it app / mission interchangeably.

hidmic, 2022-10-26:
Now that we have a wiki this is done, right @caguero ?

closed 2022-10-26 by caguero