osrf/lrauv #253: Acoustic comms improvement ideas

adityapande-1995, 2022-09-23:
I'll be using this issue ticket to discuss improvement ideas for acoustic comms plugin, to make it more realistic.

1. Signal strength information / intensity : Is it possible to supply this information via the packet ?
2. Collisions : If 2 packets from different senders reach the destination at the same time, they might not be delivered. We can add a probabilistic model for this.

adityapande-1995, 2022-10-10:
Packet collisions : https://github.com/gazebosim/gz-sim/pull/1755

adityapande-1995, 2022-11-08:
For propagation model, we could use thr model used by RFComms in gz-sim as reference.

Alternatively, I found these references : 
* https://www.mathworks.com/help/phased/ug/sonar-equation.html : Talk about signal to noise ratio and transmission loss formulae.
* https://unetstack.net/handbook/unet-handbook_modems_and_channel_models.html : UNET simulator uses BPSK fading

caguero, 2022-12-26:
See https://github.com/gazebosim/gz-sim/pull/1755 and https://github.com/gazebosim/gz-sim/pull/1793

closed 2022-12-26 by caguero