osrf/lrauv #266: Issues following installation instructions

caguero, 2022-10-26:
Before testing the LRAUV installation instructions, here's a brief summary of my system:

* I'm using Ubuntu 22.04.
* I can run gz sim locally.
* I have `nvidia-docker2` installed.

### A) Following the `Use Docker image` instructions:

After running the command:
```
xhost +
docker run --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro --gpus all osrf/lrauv:latest
```

Simulation doesn't start and I get the following error on the terminal:
```
[GUI] [Err] [Ogre2RenderEngine.cc:1100]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /home/jenkins/workspace/ogre-2.3-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
```

### B) Following the `Using a containerized workspace` instructions:
After running the command:
```
wget https://raw.githubusercontent.com/osrf/lrauv/main/tools/setup/workspace.sh -O - | bash
```

Simulation doesn't start and I get the following error on the terminal:
```
bash: line 17: BASH_SOURCE[0]: unbound variable
dirname: missing operand
Try 'dirname --help' for more information.
dirname: missing operand
Try 'dirname --help' for more information.
```

hidmic, 2022-10-26:
@caguero thanks for the review!

About A, it may have to do with your Docker version. Does changing `--gpus` for `--runtime=nvidia-docker2` have an effect?

About B, I clearly didn't test all the combinations properly. The script is too pedantic. #267 fixes it.

adityapande-1995, 2022-10-26:
Not sure if it helps, but I've been launching the simulation with ``--render-engine ogre``.

caguero, 2022-10-26:
> About A, it may have to do with your Docker version. Does changing `--gpus` for `--runtime=nvidia-docker2` have an effect?

It didn't but I found a way to make it work. I had to set the environment variable `MESA_GL_VERSION_OVERRIDE=3.3`.

Then, my command to run the sim is:
```
docker run --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro -e MESA_GL_VERSION_OVERRIDE=3.3 --gpus all osrf/lrauv:latest
```

hidmic, 2022-10-26:
@caguero you may now retry B.

caguero, 2022-10-26:
Both (A) and (B) work for me, thanks. I added to the instructions the `MESA_GL_VERSION_OVERRIDE` option.

closed 2022-10-26 by caguero