osrf/lrauv #274: Unable to find or download file (gz sim buoyant_tethys.sdf)

imasmitja, 2022-11-25:
Hi,
I installed the lrauv simulator following the instructions "Using a containerized workspace" in the Wiki. Everything went right, but when I tried to run the first example: `gz sim buoyant_tethys.sdf`   I got the following error: `Unable to find or download file`.
Nonetheless, if I execute the control commands to move the vehicle, they seem to work fine `LRAUV_example_controller` and `LRAUV_keyboard_teleop` as I can see in the terminal that the LRAUV's parameters changes.

I installed the simulator in a fresh new Ubuntu 22.04, with new Docker and nvidia-docker. I installed and ran it on a remote PC through ssh.

Do you know why I got this error?

Thanks!

arjo129, 2022-11-30:
Did you source the workspace?

imasmitja, 2022-11-30:
Yes, I did `. install/setup.bash` inside ~/lrauv_ws workspace

arjo129, 2022-11-30:
Ah OK I reallized we removed the `gz sim buoyant_tethys.sdf` example. I've updated the wiki with new instructions. Run `gz sim empty_environment.sdf`.

imasmitja, 2022-11-30:
Also, I have to say that inside the folder _~/lrauv_ws/src/_ I only have the _/lrauv_ folder not the _/lrauv-applicaion_ folder, and therefore I cannot execute the "Run the MBARI LRAUV application" commands explained in the [examples ](https://github.com/osrf/lrauv/wiki/Examples-and-tutorials). I have missed some step?

imasmitja, 2022-11-30:
> Ah OK I reallized we removed the `gz sim buoyant_tethys.sdf` example. I've updated the wiki with new instructions. Run `gz sim empty_environment.sdf`.

Ok, that looks like it works, however I'm not onsite (I'm connected to a remote machine through SSH), and I received some errors related to "qt.qpa.xcb: could not connect to display" which I think they are normal, I will test it on site later on. Thanks!

arjo129, 2022-11-30:
Ah yes we should put a header on that. The LRAUV application is no longer available as it is MBARI's own IP. @hidmic do we know how we plan on exporting the MBARI controller? If not we should remove those instructions or put a heading saying "For MBARI users only."

imasmitja, 2022-11-30:
I have an MBARI collaboration agreement which give me access to their LRAUV Bitbucket repository. In that case, what is the best case to integrate lrauv-applicaiton into ignition? should I colon the lrauv-application in the _lrauv_ws/src_ folder and build it there (inside the container I just created )? or is there another better way? how will it interact with ignition, transparently by the user? Thanks so much for your help!

arjo129, 2022-11-30:
@caguero , @hidmic , @adityapande-1995 please advice when you guys wake up. I'm sorry I don't have access to the MBARI stuff so they will be better suited to answer your question.

hidmic, 2022-12-01:
> I have an MBARI collaboration agreement which give me access to their LRAUV Bitbucket repository. In that case, what is the best case to integrate lrauv-applicaiton into ignition?

@imasmitja if you have such an agreement, then you have access to simulation usage instructions. Look for the `Tools/gz` folder.

> do we know how we plan on exporting the MBARI controller? If not we should remove those instructions or put a heading saying "For MBARI users only."

@arjo129 MBARI folks should build and push the binaries in a Docker image, like they used to. The infrastructure is there.

arjo129, 2022-12-14:
Closing this issue as it seems resolved.

closed 2022-12-14 by arjo129