osrf/lrauv #280: Help needed to use Science sensor plugin with lrauv simulation

gnanukoth, 2023-02-16:
Hello,
For a project that I am currently working on, I need the lrauv scenario simulation to test a custom control code for the robot. For the same I need to integrate the science sensors plugin to simulate relevant scenarios. I followed the installation instructions and I am able to run the empty_environment and the tethys_at_empty_environment worlds. But I would like to simulate scenarios similar to the portuguese ledge world. To get started, I tried running the tethys_at_portugueseledge world, but I am not able to. It glitches and hangs. Is there any reference world file for lrauv robot integrated with science sensors that I can begin with? Could you provide some pointers on how I can go about this? I am new to this and hence, not very confident about navigating the system, I have a doubt, I see that the other world files under the lrauv_gazebo_plugin seem to have a .em extension; Is that expected to work the same way as the other world files? May I know the difference?
Kindly guide in this regards!
Thanks!

arjo129, 2023-02-16:
Hi @gnanukoth may I know whether you are running with the mbari controller? How is gazebo installed by source or by debs? What is the error when you run it? Perhaps a log via `gz sim -v 4 tethys_at_portuuese_ledge`. By default empty environment supports science sensors. You just need to spawn an lrauv and then load your data  via the gui.

arjo129, 2023-02-16:
Hi @gnanukoth, is it possible to share the data if there is a bug we might be able to reproduce it. I believe you should be able to use the gui to load visuals using the worldconfig plugin without the need for any additional work. Are you able to see the science sensor topic?

gnanukoth, 2023-02-17:
Hello @arjo129, I am using the default data file that is provided with the repo '2003080103_mb_l3_las.csv'. My understanding was that I do not have to initialize the data file again, since I am using the default one and it is built already as part of the package. I am able to see the sensor topics, but when I do gz topic -e -t "/model/tethys/salinity" or any other science sensor topic, there is no msg received.

gnanukoth, 2023-02-17:
Regarding the portuguese_ledge, I ran the command you shared, gz sim -v 4 tethys_at_portuguese_ledge.sdf, it works fine; I am able to receive the science sensor data as well. Also when I try gz sim tethys_at_portuguese_ledge.sdf, it works fine, did not hang or get stuck today! I am not sure what the reason is. But the science sensor readings do not work with tethys_at_empty_environment.sdf. The plugins are all same and the steps followed for execution are also same. I did not initialize the data file for portuguese_ledge explicitly anywhere either. I am not sure what I am missing. Besides, the world config option in the gui is very slow and freezes before i can get to the folder where the data file is present.  
Also, when I run gz sim tethys_at_portuguese_ledge.sdf.em, it throws the following error:
Error Code 34: Msg: Invalid argument. Unable to set value [@(tiles[0].lat_deg)] for key[latitude_deg].
[Err] [Server.cc:159] Error Code 9: [/sdf/world[@name="portuguese_ledge"]/spherical_coordinates/latitude_deg:/home/anna/gazebo_models/install/lrauv_gazebo_plugins/share/lrauv_gazebo_plugins/worlds/tethys_at_portuguese_ledge.sdf.em:L85]: Msg: Error reading element <latitude_deg>
[Err] [Server.cc:159] Error Code 9: [/sdf/world[@name="portuguese_ledge"]/spherical_coordinates:/home/anna/gazebo_models/install/lrauv_gazebo_plugins/share/lrauv_gazebo_plugins/worlds/tethys_at_portuguese_ledge.sdf.em:L80]: Msg: Error reading element <spherical_coordinates>
[Err] [Server.cc:159] Error Code 9: [/sdf/world[@name="portuguese_ledge"]:/home/anna/gazebo_models/install/lrauv_gazebo_plugins/share/lrauv_gazebo_plugins/worlds/tethys_at_portuguese_ledge.sdf.em:L67]: Msg: Error reading element <world>
[Err] [Server.cc:159] Error Code 9: Msg: Error reading element <sdf>
[Err] [Server.cc:159] Error Code 1: Msg: Unable to read file:/home/anna/gazebo_models/install/lrauv_gazebo_plugins/share/lrauv_gazebo_plugins/worlds/tethys_at_portuguese_ledge.sdf.em

I am not sure if this is how I am supposed to run it. My apologies if these are very silly, I am just trying to understand how this works and how the world file has to be built, so that the file I create, works seamlessly. Thanks!

arjo129, 2023-02-17:
You are not supposed to run ` gz sim tethys_at_portuguese_ledge.sdf.em`. If
you modify the `.em` file you have to `colcon build` the workspace again.
This is because the `.em` file is actually a python empy
<https://pypi.org/project/empy/> template. The
`tethys_at_portuguese_ledge.sdf.em` is used to
generate `tethys_at_portuguese_ledge.sdf`. So if you modify
`tethys_at_portuguese_ledge.sdf.em` you should simply call `colcon build &&
gz sim tethys_at_portuguese_ledge.sdf`

> Besides, the world config option in the gui is very slow and freezes
before i can get to the folder where the data file is present.

Interesting. This may be worth investigating, however this functionality is
currently in the process of being migrated over to the mainstream gazebo.

>  My apologies if these are very silly, I am just trying to understand how
this works and how the world file has to be built, so that the file I
create, works seamlessly. Thanks!

No worries. There are no stupid questions.

gnanukoth, 2023-02-17:
Noted, Thanks! Is there any pointers for me to troubleshoot why the data is not published for the science sensors with tethys_at_empty_environment.sdf?

arjo129, 2023-02-17:
Check where you are spawning the vehicle. Is it within the field of your
data set? Portuguese ledge is located at a specific location on earth, your
data needs to be from portugues ledge's location. If you don't mind I'm
going to close this issue as it is not a bug.

arjo129, 2023-02-17:
Not a bug

closed 2023-02-17 by arjo129

gnanukoth, 2023-02-17:
Sure, Noted, Thanks!