osrf/lrauv #285: libGL error: MESA-LOADER: failed to retrieve device information

MrBlenny, 2023-09-12:
When starting in docker lrauv was crashing. 

For example (running gz gui to trigger the crash is much less noisey)

```
➜  lrauv git:(31b2b90) ✗ docker run --rm -e DISPLAY=:1 -v /tmp/.X11-unix:/tmp/.X11-unix -e MESA_GL_VERSION_OVERRIDE=3.3 --gpus all -it osrf/lrauv:latest gz gui

fixuid: fixuid should only ever be used on development systems. DO NOT USE IN PRODUCTION
fixuid: runtime UID '1000' already matches container user 'developer' UID
fixuid: runtime GID '1000' already matches container group 'developer' GID
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-developer'
libGL error: MESA-LOADER: failed to retrieve device information
intel_do_flush_locked failed: Invalid argument
```

The resolution is to mount `/dev/dri` device

eg)
```
docker run --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro -e MESA_GL_VERSION_OVERRIDE=3.3 --gpus all --device=/dev/dri:/dev/dri osrf/lrauv:latest
```

This might be a useful thing to add to the installation wiki.

marcoag, 2023-09-12:
What OS are you using? 
Are you using intel graphics?

MrBlenny, 2023-09-12:
Ubuntu Jammy with a laptop Intel CPU and NVIDIA GPU. Should be using the GPU afaik.

marcoag, 2023-09-13:
Interesting, will add this on the wiki as a side note for possible errors during installation. 

Thanks for the contribution @MrBlenny!

Closing the issue but feel free to reopen if further follow up is needed.

marcoag, 2023-09-13:
I also linked this issue on the wiki for future reference.

closed 2023-09-13 by marcoag